adafruit/Adafruit-SSD1351-library #12: Cannot convert volatile uinit32_t to PortReq on Pixel (Arduino Zero)

oconnelc, 2016-09-03:
- Arduino board:  Arduino/Genuino Zero

- Arduino IDE version: 1.6.9

I am attempting to run the pixel speed test from the Pixel kick starter: https://www.kickstarter.com/projects/rabidprototypes/pixel-the-arduino-compatible-smart-display/description 
The source is located at:  http://rabidprototypes.com/wp-content/uploads/2016/03/pixel_speedtest.zip. 

According to the specs: "The Pixel ships with the Arduino/Genuino Zero bootloader installed." 

On the Arduino IDE, I have installed:

- AdaFruit-SSD1351 V 1.0.1
- AdaFruit-GFX Library V 1.1.5. 

When I got to run, I get the following errors: 
Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Zero (Native USB Port)"

Build options changed, rebuilding all
C:\Users\Christopher\Documents\Arduino\libraries\Adafruit_SSD1351_library\Adafruit_SSD1351.cpp: In constructor 'Adafruit_SSD1351::Adafruit_SSD1351(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)':

C:\Users\Christopher\Documents\Arduino\libraries\Adafruit_SSD1351_library\Adafruit_SSD1351.cpp:465:17: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment

     csport      = portOutputRegister(digitalPinToPort(cs));

                 ^

C:\Users\Christopher\Documents\Arduino\libraries\Adafruit_SSD1351_library\Adafruit_SSD1351.cpp:468:17: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment

     rsport      = portOutputRegister(digitalPinToPort(rs));

                 ^

C:\Users\Christopher\Documents\Arduino\libraries\Adafruit_SSD1351_library\Adafruit_SSD1351.cpp:471:18: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment

     sidport      = portOutputRegister(digitalPinToPort(sid));

                  ^

C:\Users\Christopher\Documents\Arduino\libraries\Adafruit_SSD1351_library\Adafruit_SSD1351.cpp:474:19: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment

     sclkport      = portOutputRegister(digitalPinToPort(sclk));

                   ^

C:\Users\Christopher\Documents\Arduino\libraries\Adafruit_SSD1351_library\Adafruit_SSD1351.cpp: In constructor 'Adafruit_SSD1351::Adafruit_SSD1351(uint8_t, uint8_t, uint8_t)':

C:\Users\Christopher\Documents\Arduino\libraries\Adafruit_SSD1351_library\Adafruit_SSD1351.cpp:486:17: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment

     csport      = portOutputRegister(digitalPinToPort(cs));

                 ^

C:\Users\Christopher\Documents\Arduino\libraries\Adafruit_SSD1351_library\Adafruit_SSD1351.cpp:489:17: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment

     rsport      = portOutputRegister(digitalPinToPort(rs));

                 ^

I'm not sure if I'm missing a dependency, if my Arduino IDE is not supported or if the board itself is not supported.

fhervias, 2016-09-03:
I had the same issue, simply typecast the value, such that it looks something like this:
rsport      = (unsigned char*)  portOutputRegister(digitalPinToPort(rs)); 

Do this on each line the error is reported.

oconnelc, 2016-09-03:
I see the PortReq is defined by the following block: 
    #ifdef __SAM3X8E__
        typedef volatile RwReg PortReg;
        typedef uint32_t PortMask;
    #else
        typedef volatile uint8_t PortReg;
        typedef uint8_t PortMask;
    #endif

What does the _SAM3X83_ refer to? I'd be willing to write a pull request to fix the issue I found, but I want to make sure I do it right.

oconnelc, 2016-09-03:
I was able to get it work, by modifying the code from my last post to: 

```
#define __CHRIS__

#ifdef __SAM3X8E__
    typedef volatile RwReg PortReg;
    typedef uint32_t PortMask;
#elif defined __CHRIS__
    typedef volatile uint32_t PortReg;
    typedef uint32_t PortMask;
#else
    typedef volatile uint8_t PortReg;
    typedef uint8_t PortMask;
#endif
```

Is there another macro I can look for to get to the block I added? If so, I'll update the code on my side and submit a pull request.

mike-rankin, 2016-09-03:
Nice work! I wonder if getting the SSD1331 working on the SAMD21 is possible?

asurion234, 2018-03-26:
Hi, 
I'm having the same problem with the teensy 3.2 Board.
Has anyone some advice for me?
Arduino 1.8.3; Windows 7 64Bit; Teensy 3.2@48 or 96 MHz, Ucglib v. 1.5.2

C:\Users\XXXXXX\Documents\Arduino\libraries\Ucglib\src\Ucglib.cpp:795:20: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment

   u8g_data_port[0] =  portOutputRegister(digitalPinToPort(ucg->pin_list[UCG_PIN_D0]));

Best Regards, Kim

ladyada, 2019-04-13:
hiya this should be all fixed now with the SPITFT refactor (all SPI logic is up in GFX)

closed 2019-04-13 by ladyada

Rukundo-12, 2022-06-19:
rduino: 1.6.19 (Windows 11), Board: "teensy 4.0",

#include <SPI.h>

#include <Adafruit_GFX.h>
#include <Adafruit_GrayOLED.h>
#include <Adafruit_SPITFT.h>
#include <Adafruit_SPITFT_Macros.h>
#include <gfxfont.h>

#include <Adafruit_STMPE610.h>
#include <Keyboard.h>
//#include <Wire.h>  
#include <font_Arial.h>
#include <font_ArialBold.h>
#include <ILI9341_t3.h>
#include "Adafruit_MLX90640.h"         // thermal camera lib
                                     // very fast library
#include "font_Arial.h"             // fonts for the legend
// #include "Fonts\kris.h"         // fonts for the startup screen
#include <UTouch.h>                 // touchscreen lib
#define PIN_CS 10                   // chip select for the display
#define PIN_DC 9                    // d/c pin for the display

// constants for the cute little keypad
#define KEYPAD_TOP 15
#define KEYPAD_LEFT 50
#define BUTTON_W 60
#define BUTTON_H 30
#define BUTTON_SPACING_X 10
#define BUTTON_SPACING_Y 10
#define BUTTON_TEXTSIZE 2

// fire up the display using a very fast driver
ILI9341_t3 Display = ILI9341_t3(PIN_CS, PIN_DC);

// note if you have a white screen trying connecting a reset line (say pin 7 on your teensy to RST on the display and use this line
// if you have no pins left, put a 1K series resistor between Vcc and RST on the dispay and a 10 uf cap between RST and GND on the display
// basically add a low pass filter--this will slow the charge to the RST pin (you may need to experiment with the resistor and cap values
// ILI9341_t3 Display = ILI9341_t3(PIN_CS, PIN_DC, 7);

// create some colors for the keypad buttons
#define C_BLUE Display.color565(0,0,255)
#define C_RED Display.color565(255,0,0)
#define C_GREEN Display.color565(0,255,0)
#define C_WHITE Display.color565(255,255,255)
#define C_BLACK Display.color565(0,0,0)
#define C_LTGREY Display.color565(200,200,200)
#define C_DKGREY Display.color565(80,80,80)
#define C_GREY Display.color565(127,127,127)

unsigned long CurTime;

// create some text for the keypad butons
char KeyPadBtnText[12][5] = {"1", "2", "3", "4", "5", "6", "7", "8", "9", "Done", "0", "Clr" };

// define some colors for the keypad buttons
uint16_t KeyPadBtnColor[12] = {C_BLUE, C_BLUE, C_BLUE, C_BLUE, C_BLUE, C_BLUE, C_BLUE, C_BLUE, C_BLUE, C_GREEN, C_BLUE, C_RED };
uint16_t TheColor;
// start with some initial colors
uint16_t MinTemp = 25;
uint16_t MaxTemp = 35;

// variables for interpolated colors
byte red, green, blue;

// variables for row/column interpolation
byte i, j, k, row, col, incr, GetColor;
float intPoint, val, a, b, c, d, ii;
byte aLow, aHigh;
[... 552 lines cut ...]

// end of code

I tried to compile this code above but I am getting errors could any one get the following errors to help me to figure out it 
C:\Users\Hp\Downloads\arduino\libraries\UTouch\UTouch.cpp:55:8: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
  P_CLK = portOutputRegister(digitalPinToPort(T_CLK));
        ^
C:\Users\Hp\Downloads\arduino\libraries\UTouch\UTouch.cpp:57:7: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
  P_CS = portOutputRegister(digitalPinToPort(T_CS));
       ^
C:\Users\Hp\Downloads\arduino\libraries\UTouch\UTouch.cpp:59:8: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
  P_DIN = portOutputRegister(digitalPinToPort(T_DIN));
        ^
C:\Users\Hp\Downloads\arduino\libraries\UTouch\UTouch.cpp:61:9: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
  P_DOUT = portInputRegister(digitalPinToPort(T_DOUT));
         ^
C:\Users\Hp\Downloads\arduino\libraries\UTouch\UTouch.cpp:63:8: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
  P_IRQ = portInputRegister(digitalPinToPort(T_IRQ));
        ^
Multiple libraries were found for "font_Arial.h"
 Used: C:\Users\Hp\Downloads\arduino\libraries\ILI9341_t3-Library
 Not used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\ILI9341_t3
Multiple libraries were found for "Adafruit_STMPE610.h"
 Used: C:\Users\Hp\Downloads\arduino\libraries\Adafruit_STMPE610
 Not used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Adafruit_STMPE610

ladyada, 2022-06-20:
@Rukundo-12 please post on the teensy forums for help