adafruit/Adafruit-SSD1351-library #28: Arduino Due TRUE SPI not working

JakeQuinn77, 2020-03-24:
I cannot make the TEST sketch to properly display the patterns via TRUE SPI. The definitions i used are:

`#define SCLK_PIN 76
#define MOSI_PIN 75
#define DC_PIN   14
#define CS_PIN   32
#define RST_PIN  70
`
But using the same definitions but with the SOFT SPI initialization it works but very very very slow. even an arduino uno is way faster even that due even though the due is 5x faster than the uno. with a quick test on my arduino uno and due

Fill screen with black time in milliseconds(ms):

----------------------UNO------------DUE---------
SOFT SPI------------710-------------1687---------
TRUE SPI-------------63------------FAILED------

you can notice that is way slower. I can assure that the connections are good if not then the SOFT SPI would have not work. I also tried using another SPI device to check if incase the SPI pins themselves got damaged, But nope, i can communicate no problem with my other spi device.

I have tried to trace which part of the code is causing it, but it lead me to hypothesize. that maybe the library on which this library was inherited from, the GFX_library was causing it. But im not a pro at reading code. so the only red flag that i read is this

on [line 54 of  the SPITFT.h](url=https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_SPITFT.h)

`// Arduino Due?
typedef uint32_t ADAGFX_PORT_t; ///< PORT values are 32-bit
// USE_FAST_PINIO not available here (yet)...Due has a totally different
// GPIO register set and will require some changes elsewhere (e.g. in
// constructors especially).
#endif
#else                                      // !ARM
// Probably ESP8266 or ESP32. USE_FAST_PINIO is not available here (yet)
// but don't worry about it too much...the digitalWrite() implementation
// on these platforms is reasonably efficient and already RAM-resident,
// only gotcha then is no parallel connection support for now.
typedef uint32_t ADAGFX_PORT_t; ///< PORT values are 32-bit
#endif                                     // end !ARM`

_will require some changes elsewhere (e.g. in constructors especially)._ this is the only hint i got, also due uses and ARM processor. No further clues on what to change etc. I will be posting this issue on the GFX github too because it involves that library.

- Arduino board:  Due

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.12

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): Example: TEST sketch

closed 2020-03-24 by ladyada

JakeQuinn77, 2020-03-24:
I have found the problem! and it is in this library not on the GFX library. Disclaimer though i cannot speak for all the due holder since i only have a sample size of one. Here is the breakedown. 
 on the [.cpp starting at line 38](https://github.com/adafruit/Adafruit-SSD1351-library/blob/master/Adafruit_SSD1351.cpp) of this library there are definitions of default frequency with the arduino due's sam3x not being there is basically goes to the default of 24MHz which seems to be very high for the due. There are two ways to go around this. First is just lowering the default to 20MHZ (or even lower , i found this value that works with the due) or Second add a definition entry for the sam3x chip 

`#elif defined(__SAM3X8E__)

#define SPI_DEFAULT_FREQ 20000000 ///< SAM3X SPI default frequency`

although the comment said that this will be transfered to the [SPITFT.h](https://github.com/adafruit/Adafruit-GFX-Library). i have tried removing the definitions and line 227 will throw an error because SPI_DEFAULT_FREQ is not defined anymore. but it should be defined in [line 71 of SPITFT.h](https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_SPITFT.h) which is weird.

BONUS: with only fixing the frequency to 20MHZ i am getting a time of 61ms to fill the screen black. Which is good this speed is achieved by using the constructor on [line 108 of this libraries .cpp](https://github.com/adafruit/Adafruit-SSD1351-library/blob/master/Adafruit_SSD1351.cpp) and since we are not using an ESP8266 we inherit from SPITFT on line 113. but if we add another difinition after the esp8266 difinition, for the due with the same consturctor as the esp8266

`#elif defined(__SAM3X8E__)
  Adafruit_SPITFT(width, height, cs_pin, dc_pin, rst_pin) {`

The time for filling the screen black is now 60ms. not much. I dont know why is that because i have not investigated that far in

ladyada, 2020-04-02:
@JakeQuinn77 nice find, can you add a PR to the repo? https://learn.adafruit.com/an-introduction-to-collaborating-with-version-control