adafruit/Adafruit-SSD1351-library #31: setRotation() not working right

GabeHC, 2020-04-26:
setRotation not working correctly, I have to disable it.

ladyada, 2020-04-26:
hi please provide more specific information

GabeHC, 2020-04-27:
Platform : ESP32
Board : Wemos D1 R32
Project : Adafruit SSD1351 library test example.
uncomment  // tft.setRotation(1);   will screwup screen line and text,
none of any screen rotation angle will work!
```
void oled_demo(void) {
  Serial.print("hello!");
  tft.begin();

  Serial.println("init");

  // You can optionally rotate the display by running the line below.
  // Note that a value of 0 means no rotation, 1 means 90 clockwise,
  // 2 means 180 degrees clockwise, and 3 means 270 degrees clockwise.
  tft.setRotation(1);
  // NOTE: The test pattern at the start will NOT be rotated!  The code
  // for rendering the test pattern talks directly to the display and
  // ignores any rotation.
 
  uint16_t time = millis();
  tft.fillRect(0, 0, 128, 128, BLACK);
  time = millis() - time;
  
  Serial.println(time, DEC);
  delay(500);
```

Mael19, 2020-08-11:
Hello, did you manage to fix your issue ? Because I'm expecting the same with my SSD1351 screen and LOLIN D32 board. When no rotation is specified everything works fine but the I set a specific rotation all the coordinates seem to be stretched/glitched.

Mael19, 2020-08-30:
Hello, does anyone have a solution for the rotation issue ? I tried to understand, and it seems that it has something to do with the MADCTL byte, but it's too low level for me to understand right.
Help would be much appreciated!