adafruit/Adafruit-SSD1351-library #40: Display function is not working

Rishabh028, 2024-03-12:
I am trying to display the output of serial monitor in a 1.5 oled display but when i am using display.display( ); it is showing error C:\Users\Rishabh\AppData\Local\Temp.arduinoIDE-unsaved202425-13096-sx2s2m.799bi\sketch_mar5a\sketch_mar5a.ino: In function 'void loop()':
C:\Users\Rishabh\AppData\Local\Temp.arduinoIDE-unsaved202425-13096-sx2s2m.799bi\sketch_mar5a\sketch_mar5a.ino:36:13: error: 'class Adafruit_SSD1351' has no member named 'display'
display.display(); // Update the display
^~~~~~~

exit status 1

Compilation error: 'class Adafruit_SSD1351' has no member named 'display'

Janmorgen, 2024-03-20:
Hey so I had a similar problem and there isn't really a straight forward fix if you want to update the display at once with pixel data AFAIK. But I got around the issue with a combination of `tft.startWrite();` , `tft.dmaWait();`, and `tft.endWrite();`

caternuson, 2024-03-21:
It's unclear what the actual issue is here. The error message is generally explaining what is happening:
```cpp
class Adafruit_SSD1351' has no member named 'display'
```

Why is the call to `display()` being attempted? It doesn't exist.

Rishabh028, 2024-03-26:
So what should i use to display the serial monitor output of arduino ide in 1.5inch oled spi display ?

zelo66, 2024-05-27:
> Hey so I had a similar problem and there isn't really a straight forward fix if you want to update the display at once with pixel data AFAIK. But I got around the issue with a combination of `tft.startWrite();` , `tft.dmaWait();`, and `tft.endWrite();`

Can you show an example?
I don't quite understand how this can replace this function.

Rishabh028, 2024-06-25:
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1351.h>
#include <SPI.h>

#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 128

#define OLED_DC    9
#define OLED_CS    10
#define OLED_RESET 8

Adafruit_SSD1351 display = Adafruit_SSD1351(SCREEN_WIDTH, SCREEN_HEIGHT, &SPI, OLED_DC, OLED_RESET, OLED_CS);

void setup() {
  Serial.begin(9600); // initialize serial communication at 9600 baud

  display.begin(0x3D); // Initialize the OLED display
  display.fillScreen(0x0000); // Fill the screen with black
  delay(1000);
}

void loop() {
  Serial.println("Hello, world!"); // print "Hello, world!" to the serial monitor
  delay(1000); // wait for 1 second

  // Read from Serial Monitor
  if (Serial.available() > 0) {
    String serialData = Serial.readString();

    // Display on the OLED
    display.fillScreen(0x0000);  // Clear the display
    display.setCursor(0, 0);
    display.setTextSize(1);
    display.setTextColor(0xFFFF);  // White color
    display.println(serialData);
    display.display();  // Update the display
  }

  Serial.flush();  // Clear the serial buffer
  delay(100);  // Adjust delay as needed
}

Janmorgen, 2024-07-18:
Its been a while now since I made the comment, but I believe the way I used tft.startWrite(); , tft.dmaWait(), and tft.endWrite() was to use tft.dmaWait() to pause redrawing the display, then tft.startWrite before tft.drawPixel (or whatever graphical change you wanted), then I used tft.endWrite() to signal the display to update with those changes.

However, my current solution to this is to write to a canvas (GFXcanvas16), then to reference the canvas by doing something like tft.drawRGBBitmap(0,0, canvas.getBuffer(), canvas.width(),canvas.height()). This is a way more efficient way to update the canvas instantaneously (without having access to display.display())